TrakHound/MTConnect.NET #3: Issue with independent probe

NLandry91, 2017-06-07:
I am having an issue where using the Probe() method does not work, but the Start() method works fine. I have tried both Probe(baseUrl) and Probe(baseUrl, DeviceName). Would appreciate any help! Thanks!

RduschOkuma, 2017-06-07:
Could you provide an isolated example that shows your issue? It would greatly help to solve you issue.

NLandry91, 2017-06-07:
Here is an example. I used NuGet for the dependencies.

```
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;

namespace TestProbe
{
    class Program
    {
        static void Main(string[] args)
        {
            string IPAddress = "http://agent.mtconnect.org";
            var probe = new MTConnect.Clients.Probe(IPAddress);
            probe.Successful += DisplayDataItems;
            probe.Error += ErrorHandler;
            probe.ConnectionError += ConnectHandler;
            probe.Execute();
            Console.ReadLine();
        }

        static void DisplayDataItems(MTConnect.MTConnectDevices.Document document)
        {
            foreach (var device in document.Devices)
            {
                Console.WriteLine(device.Name);
            }
        }

        static void ErrorHandler(MTConnect.MTConnectError.Document errorDocument)
        {
            Console.WriteLine(errorDocument.Errors);
        }

        static void ConnectHandler(Exception e)
        {
            Console.WriteLine(e.Message);
        }
    }
}
```

RduschOkuma, 2017-06-07:
Probe.Execute() returns a MTConnect.MTConnectDevices.Document and is a synchronous call and locks the calling thread until it returns. This Function call does not cause the Probe.Successful Event to fire.

In order to use the Probe.Successful Event you need to use Probe.ExecuteAsync() which will return control to the calling method.

NLandry91, 2017-06-07:
Thanks! That fixed it.

closed 2017-06-07 by NLandry91