TrakHound/MTConnect.NET #5: NuGet 2.7.0.28439 not built from latest github code?

tekhedd, 2017-12-08:
I am seeing behavior in the NuGet 2.7.0.28439 that suggests it is built from code that does not incorporate code from the latest github trunk, dating back at least to February. 

The Probe function is not appending "/probe" to the uri. As this is apparently a change in behavior committed to trunk in Feb 2017 (for version 1.2), and it is now December, then it would appear that the NuGet code is built from a repository that is not fully in sync with the repository code. 

Just FYI

PatrickRitchie, 2017-12-11:
@tekhedd The NuGet package should be built directly using the code that is most recent on GitHub. Can you give an example of the code you are using that is not appending the "/probe" to the uri?

tekhedd, 2017-12-11:
Hmm, disassembly of the code reveals that the DLL looks correct. This is apparently actually a bug in the current release. I guess that's good? :)

            string root = "http://localhost:8123/mt";
            MTConnect.Clients.Probe probe = new MTConnect.Clients.Probe( root + "/one" );

            // at this point, probe.BaseUrl == "http://localhost:8123/mt/one";
            MTConnect.MTConnectDevices.Document doc = probe.Execute();
            Assert.IsNotNull( doc ); // doc is null

I now see what the issue is. The REST server is seeing a request for "http://localhost:8123/mt/probe", which does not return an XML response as there is no MTConnect server located at "/mt".

When it can not parse the results, Execute() simply returns null, so I was unable to trace the root of the problem. Closer examination reveals that this could be a flaw in the Uri() class. 

Hope this helps.

PatrickRitchie, 2017-12-13:
@tekhedd Thanks for the information and yes I believe this is a bug. 

You are correct that the Uri constructor discards the last element when not followed by a forward slash. It appears that the code below would fix the issue that you were seeing:

[image]

This would also need to be done for Current, Sample, and Asset requests. This fix will be added to the next version.

closed 2019-05-28 by PatrickRitchie