TrakHound/MTConnect.NET #7: DataItem attributes mapping

afennani, 2019-05-26:
in the ReadXml in
MTConnect.MTConnectStreams.DataItemCollection<T>
the copy attributes is never cleared. attributes are carried from the siblings.
there should be copy.Attributes.RemoveAll(); in the loop

PatrickRitchie, 2019-05-28:
I believe you are correct. The below should prevent the issue you described from happening should it not?

```c#
// Clear any previous Attributes
copy.Attributes.RemoveAll();

// Copy Attributes
foreach (XmlAttribute attribute in child.Attributes)
{
    var attr = doc.CreateAttribute(attribute.Name);
    attr.Value = attribute.Value;
    copy.Attributes.Append(attr);
}

// Copy Text
copy.InnerText = child.InnerText;

```

PatrickRitchie, 2022-01-31:
This is addressed and resolved in v3.0.0

closed 2022-01-31 by PatrickRitchie