TrakHound/MTConnect.NET #11: Problem with sample stream

u1alive4ever, 2020-10-12:
Hello. I apologize for my English.

I decided to share my experience of using the library. I have several devices, one of them is MTConnect 1.3, and the other is MTConnect 1.2. when using the class MTConnectClient, everything works perfectly in the case of 1.3, but there are problems with 1.2. I started getting "error in xml document (2 2)". To quickly understand this situation, I subscribed to all events output with the class and saw the error 'the interval argument to a sample request is not supported'. After that, I started searching in the source code, made sure that when calling the Start method, we really use the sample stream, but I didn't find where the error xml document (2,2) came from. Can you recommend what I should do in this case so that the code for version 1.3 and 1.2 remains the same?

Additional question. In the Namespaces class, the Clear method should completely clear the MTConnect string, leaving only <MTConnectStreams>?

Lungbo, 2020-11-19:
I Think I have the same Problem.
My Code is in VB.NET. And in the debugging Window, I can see the error "Fehler im XML-Dokument (2,2)", which actualy means the same as yours. The Configuration of the MTAgent is " MTConnect Agent 1.1.0E (MAPPS C18)". 
I can read the XML-Data in my Browser.

regards, Mo

PatrickRitchie, 2020-11-19:
I'm looking into the issue. Can you send me an example of the devices.xml file you are using for the Agent?

-Thanks!

Lungbo, 2020-12-03:
@PatrickRitchie 
Sorry, I`m late...
Here is the code of devices.xml:

[Devices.txt](https://github.com/TrakHound/MTConnect.NET/files/5634214/Devices.txt)

OskarVertetics, 2020-12-16:
I am experiencing the same or a similair problem when trying to parse a v1.3 stream, v1.4 and v1.6 works just fine. I get the error "There is an error in the XML-document" (translated to english)

Digging deeper into the exception being thrown (currently getting this on all versions but only seem to affect the application when using a v1.3 MTConnect agent):
Managed Debugging Assistant 'BindingFailure': 'Could not load the composition with the display name MTConnect-NET.XmlSerializers in the LoadFrom binding context for AppDomain with ID 1. The error was caused by: System.IO.FileNotFoundException: The file or composition could not be loaded MTConnect-NET.XmlSerializers, Version = 2.9.1.28314, Culture = neutral, PublicKeyToken = null or one of its dependencies. The file can not be found.' (translated to english)

Regards,
Oskar

Lungbo, 2020-12-30:
Hi,
I couldn´t solve the Problem. So I coded a workaround, that works well for me:

```
Private Sub Datenabruf_MTConnect(link As String)

        Dim request As WebRequest =
              WebRequest.Create(link)

        Dim response As WebResponse = request.GetResponse()

        If CType(response, HttpWebResponse).StatusDescription = "OK" Then

            Dim Maschine, counter As String
            Maschine = ""
            counter = ""

            Using dataStream As Stream = response.GetResponseStream()

                Dim reader As New StreamReader(dataStream)

                Dim responseFromServer As String = reader.ReadToEnd()

                Dim tmpSubStrings() As String = responseFromServer.Split("<")

                For Each S1 As String In tmpSubStrings

                    If S1.Contains("part_count1") Then
                        Dim S2() As String = S1.Split(">")

                        counter = S2(1)
                    End If

                    If S1.Contains("ComponentStream component=""Device"" name") Then
                        Dim S3() As String = S1.Split(""",""")
                        Maschine = S3(3)
                    End If

                Next

            End Using

            TextBox1.Text = TextBox1.Text + Now.ToShortTimeString + " Maschine: " + Maschine + " Stueckzahl: " + counter + vbCrLf

        End If

        response.Close()

    End Sub
```

PatrickRitchie, 2022-01-31:
This issue should be resoloved in v3.0.0

closed 2022-01-31 by PatrickRitchie