TrakHound/MTConnect.NET #17: Q: ShdrAdapter

MRIIOT, 2022-02-08:
Can data items be added on-the-fly after the adapter `Start` method has been called?  (Meaning, all data items do not have to be known before starting the adapter)

PatrickRitchie, 2022-02-08:
Yes DataItems can be added after the Start method is called. The Start method basically just starts the internal send queue that sends data to the Agent.

```c#
var adapter = new ShdrAdapter("device1");
adapter.Start();

var temp = new ShdrDataItem("servotemp1", 120.5);
adapter.AddDataItem(temp);
```
This will create and Start the Adapter. Then the adapter.AddDataItem(temp) will add the DataItem to the internal queue and once the queue interval elapses (default is 100ms) the SHDR line will be sent to the Agent.

If you are looking to add a DataItem to the actual Device Information Model (returned by a Probe request), you would need to embed the MTConnectAgent class and use the embedded Agent architecture. The reason you can't add edit the Device Information Model using SHDR is because the SHDR protocol doesn't have (to my knowledge) a method of sending that data to the Agent.

MRIIOT, 2022-02-08:
> If you are looking to add a DataItem to the actual Device Information Model (returned by a Probe request), you would need to embed the MTConnectAgent class and use the embedded Agent architecture. [...]

Clever.

closed 2022-02-08 by MRIIOT