TrakHound/MTConnect.NET #20: ShdrAdapter: Listen on configurable IP address

MRIIOT, 2022-03-04:
IP address to listen on should be configurable with the option to listen on all interfaces (0.0.0.0).

https://github.com/TrakHound/MTConnect.NET/blob/afb85cfdb1dd545ae04d9093039bdc4430b22bb7/src/MTConnect.NET/Adapters/Shdr/AgentClientConnectionListener.cs#L105-L113

PatrickRitchie, 2022-03-04:
You're correct this should be configurable. I will make an Address property that will default to all interfaces as you mentioned. 

I have a large commit coming up that I will most likely add early next week along with a new Release. I will include this change in with that commit.

MRIIOT, 2022-03-04:
Looking forward to it.  I'm ready to do more testing.

PatrickRitchie, 2022-03-25:
This should be fixed in 0e50b3ba0c3403b4c86f5813944cf991ec523c38

The TcpListener now listens on Any address. This may need to be configurable in the configuration file in the future.
```c#
 private async Task ClientListener(CancellationToken cancel)
        {
            do
            {
                try
                {
                    _listener = new TcpListener(IPAddress.Any, Port);
                    _listener.Start();
```

MRIIOT, 2022-03-26:
Will test in a Docker setup this week.

MRIIOT, 2022-03-28:
@PatrickRitchie , can you update Nuget package?

PatrickRitchie, 2022-03-28:
@MRIIOT Yes I will update the Nuget package and create a new Release tonight. I had a few minor other changes that I wanted to include in that Release as well.

closed 2022-03-29 by MRIIOT