TrakHound/MTConnect.NET #21: SHDR: SetUnavailable

MRIIOT, 2022-03-30:
Expected behavior: Calling `adapter.SetUnavailable` should send new SHDR to Agent with value of `UNAVAILABLE`.
Current behavior: It does not.

DataItem before setting `UNAVAILABLE`.

[image]

DataItem after setting `UNAVAILABLE`.

[image]

Comparison in `UpdateDataItem`.

[image]

[image]

PatrickRitchie, 2022-03-30:
Commit 26fc9363c8639886d5659f7b6171d76f3cdca51d should have fixed this issue.

```c#
var axesDataItem = new ShdrDataItem("f_sim_p1_axes", "X Y Z S");
adapter.AddDataItem(axesDataItem);
```
> 2022-03-30T21:02:26.7850000Z|f_sim_p1_axes|X Y Z S

```c#
adapter.SetUnavailable();
```

> 2022-03-30T21:03:17.8750000Z|f_sim_p1_axes|UNAVAILABLE

This was an issue where I was updating the instance of the ObservationInput object that was in the cached List when setting to Unavailable. This was updating the same object in the List so the "existing.ChangeId != dataItem.ChangeId" comparison was always going to be equal.

I added a few new methods for handling this that instantiate new ObservationInput objects and add them to the cached List. This will prevent the behavior described above. I also added an optional Timestamp parameter to those methods along with any other Unavailable related method.

I tested this with DataItem, Message, Condition, TimeSeries, DataSet, and Table Observations and they all seemed to work correctly.

PatrickRitchie, 2022-05-04:
Has anyone still had issues with this in the latest versions? If not, I will close this issue.

MRIIOT, 2022-05-05:
I think I was waiting for a Nuget release to test this.  But close it.

PatrickRitchie, 2022-05-05:
Ok thanks for letting me know. This update should be in any of the newest Nuget packages. 

A link to the latest package is below:
[MTConnect.NET - Nuget](https://www.nuget.org/packages/MTConnect.NET/)

closed 2022-05-05 by PatrickRitchie