TrakHound/MTConnect.NET #25: Component hierarchy from devices.xml is lost

SergeyGulik, 2022-05-23:
Hi Patrick,

At the moment when probe arrives all components are placed on the first level of hierarchy. 
[image]
So the structure is lost. I cannot figure out who is child of whom.
I found that you was trying to track hierarchy in public void AssignTypePaths(), but completely commented out this code in 0e50b3ba0c3403b4c86f5813944cf991ec523c38
Instead of just fixing component hierarchy I decided to ask what are your thoughts/ideas? How you saw it implemented?

Best regards,
Sergii

PatrickRitchie, 2022-05-23:
@SergeyGulik The Components should form a hierarchy and I haven't seen the issue of them being read as flat before.

The expected behavior is shown below where Device contains Axes & Controller and Controller contains two Paths:
[image]

The AssignTypePaths() method was from an older version of MTConnect.NET and I couldn't decide whether to leave it in or not as it shouldn't be needed (assuming everything else if working correctly) but could possibly be useful to show something like breadcrumbs in a UI. I will add this back in on the next release.

I would be curious to see what the Probe XML looks like that you read from to get the result you showed. It looks like the Components other than Axes & Controller are read as a 'generic' Component instead of a derived class based on the Type. 

-Patrick

SergeyGulik, 2022-05-27:
Hi Patrick,
I apologize. There was no bug there. All components are in place.
Best regards,
Sergii.

closed 2022-05-27 by SergeyGulik