TrakHound/MTConnect.NET #27: Multiline SHDR items are not supported

SergeyGulik, 2022-05-23:
Hi Patrick,

At the moment Assets are sent as multi-line string with serialized xml asset.
This provokes XmlException 
[image]
since ListenForAdapter handles response line by line.
I have found an idea of solution here https://github.com/mtconnect/cppagent/blob/master/README.md#assets

Best regards,
Sergii.

PatrickRitchie, 2022-05-23:
@SergeyGulik The documentation in the link you provided was what I used originally but it looks like I only implemented the first section. The "--multiline", @REMOVE_ASSET@, and @REMOVE_ALL_ASSETS@ and @UPDATE_ASSET@ commands were apparently never implemented.

I'm assuming the --multiline text is what could be causing the exception you are seeing.

I will add the missing functionality in the next release.

Thanks, 

-Patrick

PatrickRitchie, 2022-06-16:
Remove and Remove All should be working in in the [v0.5.0 release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v0.5.0-beta-agents). Let me know if anyone finds any issues with it otherwise I will close this issue.

I will look to support the Update Asset command in a later release. It should be fine in most cases to update the entire asset using the normal AddAsset() method.

SergeyGulik, 2022-06-17:
Hi Patrick,

I've tried updating to 3.4.0 nuget package and got: 
`error NU1101: Unable to find package MTConnect.NET-Services. No packages exist with this id in source(s): Microsoft Visual Studio Offline Packages, nuget.org  [.......\ALL_BUILD.vcxproj]`        
And when looking on [nuget.org](https://www.nuget.org/profiles/trakhound) there is no MTConnect.NET-Services

Best regards,
Sergii.

PatrickRitchie, 2022-06-17:
@SergeyGulik Thanks for letting me know. It looks like there was an issue and the MTConnect.NET-Services package didn't publish correctly. I just published a new version [3.4.1](https://www.nuget.org/packages/MTConnect.NET/) that should fix that dependency issue.

Thanks,

-Patrick

SergeyGulik, 2022-06-20:
Thank you, with 3.4.1 everything works fine.

closed 2022-06-20 by SergeyGulik