TrakHound/MTConnect.NET #28: Performance issue with starting adapter client in tests

PatrickRitchie, 2022-05-31:
### Discussed in https://github.com/TrakHound/MTConnect.NET/discussions/26

<div type='discussions-op-text'>

<sup>Originally posted by **SergeyGulik** May 23, 2022</sup>
Hi Patrick,

I have not put this in issues, since it not at all an issue with the code, but there exists .NET behaviour that creates me a problem.
ShdrAdapterClient.cs, line ~162
    // Create new TCP Client
    _client = new TcpClient(Server, Port);
On every test run I experienced 2sec delay on this line.
I am not a network guy at all, but luckily I have found this excellent answer on SO: https://stackoverflow.com/a/49348416
When I changed code like
    _client = new TcpClient(AddressFamily.InterNetwork); <-- this means IP4
    _client.Connect(Server, Port);
2sec delay disappeared.
This is possible solution, but according to SO answer we may also bind server to listen locally for IP6 requests.
I have not tried it yet, but wonder what would you prefer.

Best regards,
Sergii.
</div>

PatrickRitchie, 2022-05-31:
I converted this to an Issue and the delay should be fixed in commit 9d0464fd885fc389a25faff9a9531479525bf352

It looks like this could've been a combination of the delay associated with TcpClient and also possibly the ReconnectInterval configuration property not being set/used correctly.

I don't know if the delay I was seeing was from the same issue as Sergii saw but it looks like it is working faster now. There is still a delay associated with the TcpClient Timeout when trying to connect to the Adapter but this should be expected.

I set the minimum ReconnectInterval to 100ms to prevent unnecessary CPU usage but someone please let me know if you have an application where it needs to be less than this and I can change it (100ms just seemed like a reasonable minimum).

Let me know if anyone finds any issues with this otherwise I will close the issue after the upcoming release.

PatrickRitchie, 2022-06-16:
This should be fixed in in the [v0.5.0 release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v0.5.0-beta-agents). Let me know if anyone finds any issues with it otherwise I will close this issue.

SergeyGulik, 2022-06-20:
Thank you for fixing it, Patrick!

closed 2022-06-20 by PatrickRitchie