TrakHound/MTConnect.NET #36: XML Document formatted whitout newline

EcTomBot, 2022-10-22:
Hello

Mazak's MT-Connet agent version 1.5.0.14 returns an XML document on a single line (see example below).
[currentExample2.zip](https://github.com/TrakHound/MTConnect.NET/files/9843769/currentExample2.zip)

The ReadComponentStreamXml method does not read nodes correctly resulting in an exception.
Commenting on reader.Skip (), as in the example below, the procedure seems to work, but I ask for your help to correct the problem.
[image]

Thank you

PatrickRitchie, 2022-10-24:
@EcTomBot It looks like you are correct that there is an issue with reading an XML document on a single line and it looks like your proposed changes will fix it. I'm working on an update that will implement these changes and should have an update ready this week.

Thanks,
-Patrick

EcTomBot, 2022-11-08:
Hi Patrick
Have you definitively fixed the problem?
When do I assume to distribute the update?
Thank's

PatrickRitchie, 2022-11-11:
@EcTomBot Sorry for the delay in response. Yes this seems to have fixed the issue and the changes are included in the latest commit 2653770c72f9cda1ed3f58f6ee60f13563d81ca9.

I will create a new Release with this included this weekend.

Thanks for the feedback and please let me know if you find any other issues.

Thanks,
-Patrick

EcTomBot, 2022-11-11:
Thank you very much

EcTomBot, 2022-11-20:
Hi Parick
When will the new release be released?
Thanks and happy Sunday

EcTomBot, 2022-11-27:
Hi Patrick
Do you have good news for me?
Thank you

PatrickRitchie, 2022-11-28:
@EcTomBot Sorry for the delay. I just published a new [Release v1.4.0](https://github.com/TrakHound/MTConnect.NET/releases/tag/v1.4.0-agents) that includes this update.

Please let me know if you find any other issues and I appreciate the feedback.

Thanks
-Patrick

EcTomBot, 2022-11-28:
Thank you very much
Tomaso

PatrickRitchie, 2023-02-03:
@EcTomBot I just published the new [v5.0.0](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.0.0) release with this change in it.

Thanks again for your feedback and let me know if you find anything else.

Thanks,
-Patrick

closed 2024-03-04 by PatrickRitchie