TrakHound/MTConnect.NET #38: Stream reading error

SergeyGulik, 2023-01-19:
Hi Patrick,

In the changeset 9dbdbedf1d27ec27f6f6ea275c0cbd689be4c371 you introduced a problem:
[image]
You ignore the result of ReadAsync and b never get -1 as before your change.
Instead it should be something like
[image]

At least there is one more place above in the same file with the same problem.

Best regards,
Sergii Gulyk.

SergeyGulik, 2023-01-20:
It went unnoticed because usually this stream is "endless" and you stop reading it by timeout. Except a HAAS machine that does not support an "interval" parameter in _sample_ query and clearly reports it. In such case this code hangs forever aggravating memory by endlessly adding \r to the _line_.

PatrickRitchie, 2023-01-22:
@SergeyGulik Thanks for finding this and yes I believe this would prevent the issue you described if I understand the issue correctly. It looks like I misunderstood how the stream.ReadAsync() method worked and it may explain a few instances where I saw strange behavior during testing.

I'm working on an update that is available in the new [dev](https://github.com/TrakHound/MTConnect.NET/tree/dev) branch which I included this change. Let me know if this issue is fixed with that update. 

*Note that I moved the MTConnect.Clients.Rest.MTConnectClient class to MTConnect.Clients.MTConnectHttpClient*

SergeyGulik, 2023-01-23:
Hi @PatrickRitchie 
The fix looks good. When will it be released as a nuget package?
Best regards,
Sergii

PatrickRitchie, 2023-01-30:
I should be able to finish up the new Release in the next few days and have it published to Nuget

commit 301d9ae2 (2023-01-22): Update to HTTP client stream to fix #38
closed 2023-02-03 by PatrickRitchie