TrakHound/MTConnect.NET #39: ShdrQueueAdapter.AddCondition results in stack overflow

scottmwyant, 2023-01-30:
Attempting to add a condition by using a `ConditionObservationInput` object results in a Stack overflow...

```csharp
using MTConnect.Adapters.Shdr;
using MTConnect.Observations;
using MTConnect.Observations.Input;

    var adapter = new ShdrQueueAdapter("console-adapter");
    adapter.MultilineAssets = true;
    adapter.MultilineDevices = true;
    adapter.AgentConnected += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Agent Connected");
    adapter.AgentDisconnected += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Agent Disconnected");
    adapter.PingReceived += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Agent Ping Received");
    adapter.PongSent += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Pong Sent to Agent");
    adapter.LineSent += (sender, args) => Console.WriteLine($"Agent Connection (ID = {args.ClientId}) : Line Sent : {args.Message}");
    adapter.Start();

    // var c = new ConditionObservationInput("someId", ConditionLevel.NORMAL, DateTime.Now);
    var c = new ConditionObservationInput("someId");
    adapter.AddCondition(c);

```

[image]

PatrickRitchie, 2023-01-30:
Try using the "ShdrCondition" class instead of the "ConditionObservationInput" class as there seems to be an issue with using that in the AddCondition() method.

```c#
using MTConnect;
using MTConnect.Shdr;

var c = new ShdrCondition("someId");
//var c = new ShdrCondition("someId", ConditionLevel.NORMAL);
//var c = new ShdrCondition("someId", ConditionLevel.NORMAL, UnixDateTime.Now);
adapter.AddCondition(c);
```

I'm adding better documentation in the "dev" branch and should have a new Release finished this week. Below is a link to the new SHDR readme but note that there are a few changes in the new version (specifically SendCurrent() is now SendChanged()):
https://github.com/TrakHound/MTConnect.NET/tree/dev/src/MTConnect.NET-SHDR

scottmwyant, 2023-02-01:
Resolved, though there may be another problem, see #41.