TrakHound/MTConnect.NET #41: ShdrQueueAdapter.SendBuffer doesn't send ShdrCondition

scottmwyant, 2023-02-01:
The dataItem `someId` appears makes it to the agent buffer; but `someId2` fails silently.

```csharp
using MTConnect.Adapters.Shdr;
using MTConnect.Shdr;

var adapter = new ShdrQueueAdapter("console-adapter");

adapter.MultilineAssets = true;
adapter.MultilineDevices = true;
adapter.AgentConnected += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Agent Connected");
adapter.AgentDisconnected += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Agent Disconnected");
adapter.PingReceived += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Agent Ping Received");
adapter.PongSent += (sender, connectionId) => Console.WriteLine($"Agent Connection (ID = {connectionId}) : Pong Sent to Agent");
adapter.LineSent += (sender, args) => Console.WriteLine($"Agent Connection (ID = {args.ClientId}) : Line Sent : {args.Message}");

adapter.Start();

for(var i = 0; i<10; i++)
{
    adapter.AddDataItem("someId", i);
    
    var c = new ShdrCondition("someId2");
    c.Fault("This is a fault", "ABCD123");
    adapter.AddCondition(c);

    adapter.SendBuffer();
    Thread.Sleep(1500);
}
```

PatrickRitchie, 2023-02-01:
@scottmwyant It looks like the issue may be with the DeviceKey (console-adapter) not getting set for the ShdrCondition. I found a few other issues with the DeviceKey being set correctly. The ShdrCondition class contains multiple ShdrFaultState objects and their DeviceKey property was not getting set correctly when the parent ShdrCondition.DeviceKey changed.

Thanks for finding this and I will include this change in the new version.

In the meantime, you should be able to remove the "console-adapter" DeviceKey parameter in the ShdrQueueAdapter() constructor and it should work. The DeviceKey is mainly just used if an adapter is writing to multiple devices in the same Agent or if you are specifying only the DataItem Name and there are multiple devices configured on the Agent with the same Name (but different DataItemId's) and for most situations is not necessary.

## Current Output
```
Agent Connection (ID = GUTU08AOEK) : Line Sent : 2023-02-01T14:48:58.6135881Z|console-adapter:someId|9
Agent Connection (ID = GUTU08AOEK) : Line Sent : 2023-02-01T14:48:58.6135881Z|someId2|FAULT|ABCD123|||This is a fault
```

## Output after changes
```
Agent Connection (ID = CIEJM8ZSKK) : Line Sent : 2023-02-01T14:50:24.5075676Z|console-adapter:someId|1
Agent Connection (ID = CIEJM8ZSKK) : Line Sent : 2023-02-01T14:50:24.5079585Z|console-adapter:someId2|FAULT|ABCD123|||This is a fault
```

scottmwyant, 2023-02-01:
Thanks for the workaround; conditions are getting to the agent when `deviceKey` is not used in the constructor.

closed 2023-02-01 by scottmwyant