TrakHound/MTConnect.NET #42: Using the MQTT-Relay agent to send data to MQTT Broker with client certificate/key

benaish, 2023-02-19:
I'm using the MQTT-Relay agent to read massages from a device (SHDR) and publish them to Greengrass' MQTT Broker (moquette), in order for them to end up in AWS IoT Core. Since there's a client authentication, the massage publishing is failing due to an authentication error. Is there a way to configure the client side certificates/key?

I couldn't find an option to work with Client certificate and a private key also by changing the code, but I did try the following code change to include client pfx certificate with a password. Running this resulted in an error as you can see below.
Any idea or suggestion how to enable working against the Greengrass MQTT?

```
                            if (string.IsNullOrEmpty(_configuration.MqttCert) || string.IsNullOrEmpty(_configuration.MqttCaCert))
                            {
                                mqttClientOptions = new MqttClientOptionsBuilder()
                                .WithTcpServer(_configuration.Server, _configuration.Port)
                                .Build();
                            }
                            else
                            {
                                var caCert = new X509Certificate2(_configuration.MqttCaCert);
                                var clientCert = new X509Certificate2(_configuration.MqttCert, _configuration.MqttCertPwd); 

                                mqttClientOptions = new MqttClientOptionsBuilder()
                                .WithTcpServer(_configuration.Server, _configuration.Port)
                                .WithClientId(_configuration.MqttClientId)
                                .WithTls(new MqttClientOptionsBuilderTlsParameters
                                {
                                    UseTls = true,
                                    //AllowUntrustedCertificates = true,
                                    SslProtocol = System.Security.Authentication.SslProtocols.Tls12,
                                    Certificates = new List<X509Certificate> { caCert, clientCert }                                   
                                    
                                })

                                .Build();
                            }
                        }
```

<code>2023-02-19T15:47:12.113Z [INFO] (nioEventLoopGroup-17-6) io.moquette.broker.metrics.MQTTMessageLogger: Channel Inactive. {}
2023-02-19T15:47:18.024Z [ERROR] (nioEventLoopGroup-17-7) io.moquette.broker.NewNettyMQTTHandler: Unexpected exception while processing MQTT message. Closing Netty channel. CId=null. {}
io.netty.handler.codec.DecoderException: javax.net.ssl.SSLHandshakeException: Received fatal alert: certificate_unknown
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:480)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:279)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.netty.handler.timeout.IdleStateHandler.channelRead(IdleStateHandler.java:286)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.moquette.broker.metrics.BytesMetricsHandler.channelRead(BytesMetricsHandler.java:51)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:722)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:584)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:995)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: javax.net.ssl.SSLHandshakeException: Received fatal alert: certificate_unknown
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:131)
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:117)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:340)
	at java.base/sun.security.ssl.Alert$AlertConsumer.consume(Alert.java:293)
	at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:186)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:172)
	at java.base/sun.security.ssl.SSLEngineImpl.decode(SSLEngineImpl.java:681)
	at java.base/sun.security.ssl.SSLEngineImpl.readRecord(SSLEngineImpl.java:636)
	at java.base/sun.security.ssl.SSLEngineImpl.unwrap(SSLEngineImpl.java:454)
	at java.base/sun.security.ssl.SSLEngineImpl.unwrap(SSLEngineImpl.java:433)
	at java.base/javax.net.ssl.SSLEngine.unwrap(SSLEngine.java:637)
	at io.netty.handler.ssl.SslHandler$SslEngineType$3.unwrap(SslHandler.java:295)
	at io.netty.handler.ssl.SslHandler.unwrap(SslHandler.java:1342)
	at io.netty.handler.ssl.SslHandler.decodeJdkCompatible(SslHandler.java:1235)
	at io.netty.handler.ssl.SslHandler.decode(SslHandler.java:1284)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:510)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:449)
	... 25 more
</code>

PatrickRitchie, 2023-02-20:
@benaish Thanks for letting me know about the issue and I started looking into this and it seems that you are on the right track.

With some changes, I got the Relay Agent to connect to an instance of AWS IoT that I setup. The next issue is that AWS IoT apparently only supports 7 forward slashes in the topic names. So I will need to either rethink the existing topic format or add a configuration option to toggle between "flat" and "hierarchical".

I'll have an update (along with some documentation) that I will push to the **dev** branch with the configuration and support for certificates. It looks something like below:

```c#
// Declare new MQTT Client Options with Tcp Server
var clientOptionsBuilder = new MqttClientOptionsBuilder().WithTcpServer(_configuration.Server, _configuration.Port);

var certificates = new List<X509Certificate2>();

// Add CA (Certificate Authority)
if (!string.IsNullOrEmpty(_configuration.CertificateAuthority))
{
    certificates.Add(new X509Certificate2(GetFilePath(_configuration.CertificateAuthority)));
}

// Add Client Certificate & Private Key
if (!string.IsNullOrEmpty(_configuration.PemClientCertificate) && !string.IsNullOrEmpty(_configuration.PemPrivateKey))
{
    certificates.Add(new X509Certificate2(X509Certificate2.CreateFromPemFile(GetFilePath(_configuration.PemClientCertificate), GetFilePath(_configuration.PemPrivateKey)).Export(X509ContentType.Pfx)));

    clientOptionsBuilder.WithCleanSession();
    clientOptionsBuilder.WithTls(new MqttClientOptionsBuilderTlsParameters()
    {
        UseTls = true,
        SslProtocol = System.Security.Authentication.SslProtocols.Tls12,
        IgnoreCertificateRevocationErrors = true,
        IgnoreCertificateChainErrors = true,
        AllowUntrustedCertificates = true,
        Certificates = certificates
    });
}

// Add Credentials
if (!string.IsNullOrEmpty(_configuration.Username) && !string.IsNullOrEmpty(_configuration.Password))
{
    if (_configuration.UseTls)
    {
        clientOptionsBuilder.WithCredentials(_configuration.Username, _configuration.Password).WithTls();
    }
    else
    {
        clientOptionsBuilder.WithCredentials(_configuration.Username, _configuration.Password);
    }
}

// Build MQTT Client Options
var clientOptions = clientOptionsBuilder.Build();

await _mqttClient.ConnectAsync(clientOptions, CancellationToken.None);
```

PatrickRitchie, 2023-02-22:
@benaish I pushed some updates to the [dev](https://github.com/TrakHound/MTConnect.NET/tree/dev) branch that should allow you to test a connection to AWS IoT.

I also added instructions with connection information specific to AWS IoT and an example of what the configuration should look like:

https://github.com/TrakHound/MTConnect.NET/blob/dev/docs/MQTT-AWS-IoT.md

Please let me know if you have any issues or questions otherwise I will add this to the main branch and create a new release within the next week.

Thanks,
-Patrick

benaish, 2023-02-28:
> Please let me know if you have any issues or questions otherwise I will add this to the main branch and create a new release within the next week.

Thanks Patrick, I will try it out!
By the way - did you test it directly with AWS IoT, or also through a local Greengrass MQTT broker?

Thanks,
Moshe

PatrickRitchie, 2023-03-03:
Moshe, I only tested it directly with AWS IoT. I'm working on a test with Greengrass but I'm assuming it should be the same or similar.

I just published a new [v5.1.0 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.1.0) that includes these changes and an updated version of the [AWS IoT Guide](https://github.com/TrakHound/MTConnect.NET/blob/master/docs/MQTT-AWS-IoT.md).

Please let me know if you have any other questions

Thanks,
-Patrick

benaish, 2023-03-05:
> Moshe, I only tested it directly with AWS IoT. I'm working on a test with Greengrass but I'm assuming it should be the same or similar.
>
> I just published a new [v5.1.0 [...]

Thanks Patrick! 
I have tested it against Greengrass and directly against IoT Core and ran into some issues:

For **Greengrass** - I'm unable to pass the authentication - 
1. I noticed that there's no configuration for Client ID - I used clientOptionsBuilder.WithClientID for testing (it would be good to have it as part of the agent.cfg)
2. Even after testing it with the client ID i'm getting an error saying Certificate unkown. I was able to use the same certificate/key with a C++ MTConnect agent and it worked, so i'm quite sure the certificates i'm using are fine.

Working directly against **AWS IoT**:
I'm able to connect to AWS IoT MQTT Broker, but I'm unable to publish messages to it.
I'm getting an Exception (OperationCanceledException) while publishing messages, but I'm not sure why. 
StackTrace:    
```
at MQTTnet.Adapter.MqttChannelAdapter.WrapAndThrowException(Exception exception)
   at MQTTnet.Adapter.MqttChannelAdapter.<SendPacketAsync>d__32.MoveNext()
   at MQTTnet.Client.MqttClient.<PublishAtMostOnce>d__70.MoveNext()
   at MTConnect.Mqtt.MTConnectMqttRelay.<Publish>d__56.MoveNext() in ...\MTConnect.NET\src\MTConnect.NET-MQTT\MTConnectMqttRelay.cs:line 281
```

Any help here is much appreciated!

Thanks,
Moshe

PatrickRitchie, 2023-03-05:
Thanks for the information. I will add the ClientId parameter to the Agent and Client configurations.

I'm in the process of getting a test environment setup for Greengrass so I should be able to find out why the certificates aren't working.

For working directly against AWS IoT, make sure that you have the **mqttFormat: Flat** parameter is set in the **agent.config.yaml** file. I was getting the same OperationCanceledException when trying to publish to a Topic with more than 7 forward slashes which is apparently a limit in AWS IoT.

PatrickRitchie, 2023-03-05:
Moshe,

I just published a new [5.2.0 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.2.0) that includes the ClientId configuration parameter.

I'm still trying to get Greengrass setup to test but I'll let you know what I find that could be the issue.

Thanks,
-Patrick

PatrickRitchie, 2023-03-08:
Moshe,

Are you able to connect to Greengrass using [MQTT Explorer](https://mqtt-explorer.com/)? Just curious whether you can connect using that app and what the configuration for that would be so we could mimic it in the Agent config.

There is a configuration option **allowUntrustedCertificates** that can be set to **true** and I'm curious if you have tried that.
```yaml
# The hostname of the MQTT broker to publish messages to
server: localhost

# # The port number of the MQTT broker to publish messages to
port: 8883

certificateAuthority: C:\greengrass\v2\rootCA.pem
pemCertificate: C:\Users\patrick\Downloads\321654657498791431546-certificate.pem.crt
pemPrivateKey: C:\Users\patrick\Downloads\123654987984651498486-private.pem.key

mqttFormat: Flat
retainMessage: false
allowUntrustedCertificates: true
clientId: trakhound-test
```

benaish, 2023-03-08:
I'm able to connect and send messages using MQTT Explorer. using the following configurations:
Encryption: On
Validate certificate: Off
Server hostname: MyServer1
Port: 8883
Server certificate - ca.pem, myClient.crt and key. 
ClientID: MyClientDevice

Just tried `allowUntrustedCertificates: true
`
I'm getting an immediate timeout error when executing `_mqttClient.ConnectAsync`: 
`MQTTnet.Adapter.MqttConnectingFailedException: 'Error while authenticating. The operation has timed out.'`

benaish, 2023-03-13:
> I'm able to connect and send messages using MQTT Explorer. using the following configurations: Encryption: On Validate certificate: Off Server hostname: MyServer1 #this one is a greengrass server Port: 8883 Server certificate - ca.pem, myClient.crt and key. ClientID: MyClientDevice
> 
> Just tried `allowUntrustedCertificates: true
> 
> I'm now getting the certificate unknown error on the greengrass side: `2023-03-12T07:41:46.971Z [INFO] (nioEventLoopGroup-3-5) io.moquette.broker.metrics.MQTTMessageLogger: Channel Inactive. {} 2023-03-12T07:43:10.588Z [ERROR] (nioEventLoopGroup-3-6) io.moquette.broker.NewNettyMQTTHandler: Unexpected exception while processing MQTT message. Closing Netty channel. CId=null. {} io.netty.handler.codec.DecoderException: javax.net.ssl.SSLHandshakeException: Received fatal alert: certificate_unknown`
> 
> And this on the client side: `MQTTnet.Exceptions.MqttCommunicationException: 'The remote certificate was rejected by the provided RemoteCertificateValidationCallback.`

**Update**
I was able to work around the issue by skipping the client side validation - created a CertificateValidationHandler that returns true.
I believe that my greengrass server certificate was created with localhost instead of the actual hostname.

Thanks for the help!

benaish, 2023-03-13:
Hi @PatrickRitchie,
Is it possible to include additional information in the payload of the MQTT message? 
For example: Category (event/sample), since i'm using the flat mode and this info is no longer part of the topic. Perhaps it's also  possible to include additional info like units?

Thanks,
Moshe

PatrickRitchie, 2023-03-13:
@benaish Thanks for the update. I was able to test connecting to Greengrass successfully myself and I documented what worked for me [Here](https://github.com/TrakHound/MTConnect.NET/blob/master/docs/MQTT-AWS-Greengrass-Moquette.md). I saw the CertificateValidationHandler function that could be used but I wanted to read more information about it and make sure that is not defeating the purpose of using a certificate at all.

I will look into adding additional information to the MQTT messages. I was able to get any additional information by subscrbing to the **Device** topic (ex. MTConnect/Devices/OKUMA.Lathe.123456/Device). There could be a lot of additional information associated with a DataItem so I am cautious to only include some, and not others, in the Observation payload.

An example of a client that mimics the MTConnect REST protocol is the MTConnectMqttClient class and can be found [Here](https://github.com/TrakHound/MTConnect.NET/blob/master/src/MTConnect.NET-MQTT/Clients/MTConnectMqttClient.cs). It will subscribe to the Device (ex. MTConnect/Devices/OKUMA.Lathe.123456/#) and then parse the messages that it receives based on whether the topic is for Device Model, Assets, or Observations. This would be where I would read the additional metadata for each DataItem by reading from the DeviceReceived event in the MTConnectMqttClient class.

Please let me know if you have any questions about how to implement this.

PatrickRitchie, 2023-03-15:
@benaish I published the [v5.3.0 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.3.0) earlier today that should address this issue. 

Please let me know if you find any other issues or have any other questions.

Thanks,
-Patrick

closed 2023-03-15 by benaish