TrakHound/MTConnect.NET #44: Cultute specific XML serialization

SergeyGulik, 2023-03-03:
Hi Patrick,
I have found one funny error in Xml serialization.
```
public class XmlFilter
{
....
        public static void WriteXml(XmlWriter writer, IFilter filter)
        {
            if (filter != null)
            {
                writer.WriteStartElement("Filter");
                writer.WriteAttributeString("type", filter.Type.ToString());
                writer.WriteString(filter.Value.ToString());<------------------
                writer.WriteEndElement();
            }
        }
```
Please change pointed line to 
`writer.WriteValue(filter.Value);`
Otherwise, double value is serialized according to the local system settings. 
In my case 0.5 is serialized as 0,5 :) producing invalid xml.

Best regards,
Sergii Gulyk.

PatrickRitchie, 2023-03-05:
Sergii, thanks for finding this and yes I will add this change.

SergeyGulik, 2023-03-05:
Patrick, thank you so much for a prompt fix!
Is there any forecast when it will be released as the nuget package?

PatrickRitchie, 2023-03-05:
This was pretty easy to test so I went ahead and published a new release [5.2.0](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.2.0) that should be available on Nuget as well.

Let me know if you find anything else or have any other questions.

Thanks,
-Patrick

SergeyGulik, 2023-03-06:
Thank you! Applied the fix.

closed 2023-03-06 by SergeyGulik