TrakHound/MTConnect.NET #45: Receiving Category information with Flat topic mode

benaish, 2023-03-15:
I'm using the Flat topic mode due to level limitation in AWS IoT Core.
By doing that I'm losing important information that i would like to get as part of each message - Category (Event/Sample/..).  It would be great to keep getting this information also on the flat topic mode. Another option could be to get this as part of the payload - but currently the Json formatter skips the Category when creating the payload out of the observation:
`return CreateMessage(topic, Formatters.EntityFormatter.Format(documentFormatterId, observation));`

Related to a discussion in issue #42

PatrickRitchie, 2023-03-16:
@benaish I agree that each message should contain Category. I'm looking at how to handle the MQTT Json to contain Category but the regular Json (sent over Http) to not contain Category since it is part of a full Response Document as opposed to MQTT that only sends the Entity.

My approach will most likely be to use an Options list (KeyValuePair) to pass a flag to the JsonEntityFormatter.Format(IEnumerable<IObservation> observations). I have a similar Options list for other methods in the EntityFormatter class.

I should have this added in a new release over the weekend.

Let me know if you have any further thoughts on how to handle this or if you have any other questions.

Thanks,
-Patrick

PatrickRitchie, 2023-03-20:
I just published the new [5.4.0 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.4.0) that contains this fix as well as some other additions to the MQTT implementation. More information can be found [Here](https://github.com/TrakHound/MTConnect.NET/blob/master/docs/MQTT-Protocol.md).

Let me know if this is what you were looking for.

Thanks,
-Patrick

benaish, 2023-03-21:
> have a similar Options list for other methods in the EntityFormatter class.

That's great, I will look into it!

Thanks,
Moshe

closed 2023-03-22 by benaish