TrakHound/MTConnect.NET #46: ShdrIntervalQueueAdapter ShdrCondition Fault vs AddFault behavior.

MRIIOT, 2023-03-28:
NuGet release 5.0.0.

```
void SetCondition(key, isFault, nativeCode, text)
{
  var c = new ShdrCondition(key);
  if(isFault) {
    c.AddFault(text, nativeCode);
  } else {
    c.AddNormal(text, nativeCode);
  }
}
```

Using above approach to set the condition, if I send a fault, fault, normal, normal, then all four instances register as samples at the cppagent.  

```
2023-03-28T19:22:00Z|f_sim_p1_alm_logic|FAULT|SW0100|||PARAMETER ENABLE SWITCH ON
2023-03-28T19:22:01Z|f_sim_p1_alm_logic|FAULT|SW0100|||PARAMETER ENABLE SWITCH ON
2023-03-28T19:22:02Z|f_sim_p1_alm_logic|NORMAL|SW0100|||PARAMETER ENABLE SWITCH ON
2023-03-28T19:22:03Z|f_sim_p1_alm_logic|NORMAL|SW0100|||PARAMETER ENABLE SWITCH ON
```

Is it expected for the SHDR adapter to send conditions through even if they have not changed?

```
<Fault dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1327" timestamp="2023-03-28T19:21:06.187136Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Fault>
<Fault dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1329" timestamp="2023-03-28T19:21:07.181688Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Fault>
<Fault dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1331" timestamp="2023-03-28T19:21:08.19261Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Fault>
<Fault dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1333" timestamp="2023-03-28T19:21:09.190217Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Fault>
<Normal dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1336" timestamp="2023-03-28T19:21:10.159083Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Normal>
<Normal dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1339" timestamp="2023-03-28T19:21:11.188217Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Normal>
<Normal dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1341" timestamp="2023-03-28T19:21:12.192046Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Normal>
<Normal dataItemId="f_sim_p1_alm_logic" nativeCode="SW0100" sequence="1343" timestamp="2023-03-28T19:21:13.207794Z" type="LOGIC_PROGRAM">PARAMETER ENABLE SWITCH ONrh</Normal>
```

PatrickRitchie, 2023-03-28:
Try setting OutputTimestamps = false for the ShdrAdapter. 

#### Example:
```c#
var adapter = new ShdrIntervalAdapter();
adapter.OutputTimestamps = false;
```

Otherwise, if you don't set the timestamp explicitly then a new timestamp will get set for the Condition Fault State and it will be read as unique and be sent to the Agent.

Let me know if that fixes the issue or not.

MRIIOT, 2023-03-28:
Disabling OutputTimestamp does omit timestamp from being sent with SHDR but also causes cppagent 2.1.0.6 to throw an error.

```
 [error] Connector::reader->Connector::parseSocketBuffer->Connector::processLine->ShdrAdapter::processData: Error in processData: invalid stod argument
```

PatrickRitchie, 2023-03-28:
Do you have the raw SHDR output? It should be valid if it follows the format below:

```
f_sim_p1_alm_logic|FAULT|SW0100|||PARAMETER ENABLE SWITCH ON
f_sim_p1_alm_logic|NORMAL|SW0100|||PARAMETER ENABLE SWITCH ON
```

MRIIOT, 2023-03-28:
```
f_sim_p1_alm_logic|FAULT|SW0100|||PARAMETER ENABLE SWITCH ON
f_sim_p1_alm_logic|NORMAL|SW0100|||PARAMETER ENABLE SWITCH ON
```

PatrickRitchie, 2023-03-28:
I may have told you wrong on my first response. It should filter out duplicates regardless of the Timestamp. This was related to Issue #40. 

It seems to be filtering properly testing with cppagent 2.0.0.11:
[image]

Are you using the ShdrIntervalAdapter class? Or are you manually sending using the SendChanged() or SendLast() methods?

You should be able to set OutputTimestamps back to True.

However, you did discover a new issue which is that the Agent apparently expects a preceding "|" character for a SHDR line without a timestamp. I will upload a fix for this tonight.

MRIIOT, 2023-03-28:
I do no call SendChanged or SendLast, I let the adapter handle sending.

Trace output from cppagent.

```
2023-03-28T20:50:38.215961Z (0x00001c1c) [trace] Connector::reader->Connector::parseSocketBuffer: (127.0.0.1:7878) 94 characters in incomming buffer
2023-03-28T20:50:38.215961Z (0x00001c1c) [trace] Connector::reader->Connector::parseSocketBuffer->Connector::processLine: (127.0.0.1:7878) Received line: f_sim_adapter_health|NORMAL||||
2023-03-28T20:50:38.215961Z (0x00001c1c) [error] Connector::reader->Connector::parseSocketBuffer->Connector::processLine->ShdrAdapter::processData: Error in processData: invalid stod argument
2023-03-28T20:50:38.215961Z (0x00001c1c) [trace] Connector::reader->Connector::parseSocketBuffer: (127.0.0.1:7878) 62 characters in incomming buffer
2023-03-28T20:50:38.215961Z (0x00001c1c) [trace] Connector::reader->Connector::parseSocketBuffer->Connector::processLine: (127.0.0.1:7878) Received line: f_sim_p1_alm_logic|NORMAL|SW0100|||PARAMETER ENABLE SWITCH ON
2023-03-28T20:50:38.215961Z (0x00001c1c) [error] Connector::reader->Connector::parseSocketBuffer->Connector::processLine->ShdrAdapter::processData: Error in processData: invalid stod argument
```

PatrickRitchie, 2023-03-28:
Ok sounds like it was just an issue with the SHDR output.

I just published the [v5.4.1 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v5.4.1) (and also on Nuget) that should fix this issue with OutputTimestamps = false. 

See if that fixes the issue and let me know if you run into anything else.

Thanks,
-Patrick

MRIIOT, 2023-03-29:
5.4.1 did fix sending SHDR without timestamp by prepending a pipe.

Below does filter duplicates.

```
_adapter.FilterDuplicates = true;

void SetCondition(key, isFault, nativeCode, text)
{
  var c = new ShdrCondition(key);
  if(isFault) {
    c.AddFault(text, nativeCode);
  } else {
    c.AddNormal(text, nativeCode);
  }
  _adapter.AddCondition(c);
}
```

closed 2023-03-29 by MRIIOT

PatrickRitchie, 2023-03-29:
Glad to hear it is working now. Let me know if you come across anything else.