TrakHound/MTConnect.NET #50: HTTP field from client is too long - 414

ynad, 2023-06-26:
Hello,

I'm experiencing issues on a client which is using your assemblies to read from MTConnect sources.
I'm not 100% sure this will be the right place to ask this, but after 1 month of troubleshooting I'm almost out of options.

The requests seem to fail because the client is sending HTTP headers too long.

This is the error on the client, which is using MTConnect 2.0:
````
11:16:14.772 Driver Interface: xxx Connection Error Exception: Response status code does not indicate success: 414 (HTTP field from client is too long).
````

This is on one of the involved MTConnect agents, configured for 2 machines.
The agent used is: [https://github.com/mtconnect/cppagent](cppagent). Tried different versions, 1.4, 1.8, 2.1, but XML schemas are still 1.3 because provided by machine manufacturer (and has not released any new one yet...). If necessary I can post XML schemas involved.
Version 1.4/1.8 logs:
````
2023-06-06T09:39:48.291499Z: ERROR [12] HttpServer: Error processing request from: 127.0.0.1 - HTTP field from client is too long
2023-06-06T09:39:48.291716Z: DEBUG [12] agent: Returning error INVALID_REQUEST: Error processing request from: 127.0.0.1 - HTTP field from client is too long
````
Version 2.1 logs:
```
2023-06-26T14:02:00.093997Z (0x00000e48) [debug] Connector::heartbeat: Sending heartbeat
2023-06-26T14:02:00.093997Z (0x00000e48) [debug] Connector::heartbeat->Connector::sendCommand: (Port:50838) Sending PING
2023-06-26T14:02:00.093997Z (0x00000e48) [debug] Connector::reader->Connector::parseSocketBuffer->Connector::processLine: (Port:50838) Received a PONG for 10.xx.xx.xx on port 7878
2023-06-26T14:02:09.359244Z (0x00000e48) [info] SessionImpl::requested: ReST Request: From [127.0.0.1:50855]: GET /probe
2023-06-26T14:02:09.378237Z (0x00000e48) [info] SessionImpl::requested: ReST Request: From [127.0.0.1:50856]: GET /current
2023-06-26T14:02:09.411237Z (0x00000e48) [warning] SessionImpl::requested->Session::fail: Operation failed: Could not read request
2023-06-26T14:02:09.411237Z (0x00000e48) [warning] SessionImpl::requested->Session::fail: Closing: header limit exceeded - header limit exceeded
```

Following is a longer extract of client logs. 
At first values are read, then a connection exception is thrown. This happens at every data polling from MTConnect source, be it every 1 second or every 30s.

````
15:00:15.010 DevicesSuccessful: Driver Interface: xxx

15:00:15.037 MTConnect Data Change Values:
Spindle_Override 93
EStop_State ARMED
avail AVAILABLE
dev_asset_chg 
dev_asset_chg UNAVAILABLE
dev_asset_rem 
dev_asset_rem UNAVAILABLE
Block_Number 0
Current_Tool 0
Feed_Override 79
Part_Count 2325
Program_Name_Editing 89552-TRATT.HWM
Program_Status READY
Rapid_Override 100
Spindle_Speed 0
Feed_Rate 0
Program_Runtime_Seconds 0
Spindle_Time 0

15:00:15.100 Driver Interface: xxx Connection Error Exception: Response status code does not indicate success: 414 (HTTP field from client is too long).
15:00:24.431 MTConnect Data Change Values:
servo UNAVAILABLE
spndl UNAVAILABLE
at UNAVAILABLE
ct UNAVAILABLE
spc UNAVAILABLE
tmp UNAVAILABLE
ct2 UNAVAILABLE
spc2 UNAVAILABLE
tmp2 UNAVAILABLE
ccond UNAVAILABLE
logic UNAVAILABLE
system UNAVAILABLE
coolhealth UNAVAILABLE
electric UNAVAILABLE
hydhealth UNAVAILABLE
lube UNAVAILABLE
motion UNAVAILABLE
path_system UNAVAILABLE
pneucond UNAVAILABLE
xt UNAVAILABLE
yt UNAVAILABLE
zt UNAVAILABLE
aaxisstate UNAVAILABLE
arf UNAVAILABLE
caxisstate UNAVAILABLE
rf UNAVAILABLE
c2axisstate UNAVAILABLE
rf2 UNAVAILABLE
estop UNAVAILABLE
pltnum UNAVAILABLE
avail UNAVAILABLE
d1_asset_chg 
d1_asset_chg UNAVAILABLE
d1_asset_rem 
d1_asset_rem UNAVAILABLE
functionalmode UNAVAILABLE
door UNAVAILABLE
Sovr UNAVAILABLE
exec UNAVAILABLE
hd1chuckstate UNAVAILABLE
ln UNAVAILABLE
mode UNAVAILABLE
pc UNAVAILABLE
pcmt UNAVAILABLE
peditmode UNAVAILABLE
peditname UNAVAILABLE
pfo UNAVAILABLE
pfr UNAVAILABLE
pgm UNAVAILABLE
seq urn:mazakusa.com:MazakStreams:1.3
seq UNAVAILABLE
spcmt UNAVAILABLE
spgm UNAVAILABLE
tid UNAVAILABLE
tid2 urn:mazakusa.com:MazakStreams:1.3
tid2 UNAVAILABLE
tid3 urn:mazakusa.com:MazakStreams:1.3
tid3 UNAVAILABLE
unit urn:mazakusa.com:MazakStreams:1.3
unit UNAVAILABLE
xaxisstate UNAVAILABLE
yaxisstate UNAVAILABLE
zaxisstate UNAVAILABLE
af UNAVAILABLE
al UNAVAILABLE
aposm UNAVAILABLE
aposw UNAVAILABLE
cf UNAVAILABLE
cl UNAVAILABLE
cposm UNAVAILABLE
cposw UNAVAILABLE
cs UNAVAILABLE
ctemp UNAVAILABLE
sl UNAVAILABLE
cf2 UNAVAILABLE
cl2 UNAVAILABLE
cpos2m UNAVAILABLE
cpos2w UNAVAILABLE
cs2 UNAVAILABLE
ctemp2 UNAVAILABLE
sl2 UNAVAILABLE
atime UNAVAILABLE
ctime UNAVAILABLE
tcltime UNAVAILABLE
yltime UNAVAILABLE
concentration UNAVAILABLE
cooltemp UNAVAILABLE
pf UNAVAILABLE
xf UNAVAILABLE
xl UNAVAILABLE
xpm UNAVAILABLE
xpw UNAVAILABLE
yf UNAVAILABLE
yl UNAVAILABLE
ypm UNAVAILABLE
ypw UNAVAILABLE
zf UNAVAILABLE
zl UNAVAILABLE
zpm UNAVAILABLE
zpw UNAVAILABLE
````

Is there any reason I didn't find which may cause these connection errors? Is it because of the older version of MTConnect schema (1.3), and some incompatibilities? 
Which I could perfectly understand, but sadly I cannot change some agent and schema versions, they are either built-in in the machines or not customizable/upgradable by me, end customer.

Any help will be very appreciated. If I missed some details or more info is needed please let me know.

Thank!

PatrickRitchie, 2023-06-26:
Thanks for the information. Are you able to open the MTConnect data in a web browser without the error? 

ex. http://[Agent_Address]:5000/probe or http://[Agent_Address]:5000/current

Based on the error, I would think the issue should be on the Agent side as the client should only be sending the URL with no HTTP request body for both a probe and current request.

ynad, 2023-06-27:
> Thanks for the information. Are you able to open the MTConnect data in a web browser without the error?
>
> ex. http://[Agent_Address]:5000/probe or http://[Agent_Address]:5000/current
>
> Based on the [...]

Hello,
thanks for your reply.

Yes, I'm able to read MTConnect data via web browser, on both agents for current and probe requests.
Below a couple of example output.

First agent, locally installed (atm running version 1.8, but can upgrade to 2.0), configured to connect to 2 machines of the same manufacturer:
[image]

Second agent built-in on the machine, different manufacturer:
[image]

PatrickRitchie, 2023-06-27:
Thanks for the information. The first thing that I can think of would be that the ContentEncodings header is being sent and not expected by the Agent.

You should be able to disable sending that header by setting the **ContentEncodings** property on the MTConnectHttpClient to null.

```c#
var client = new MTConnectHttpClient("localhost", 5000);
client.ContentEncodings = null;
```

Let me know if that fixes the issue.

PatrickRitchie, 2023-07-04:
Glad to hear it is working and please let me know if you run into any other issues.

closed 2024-03-04 by PatrickRitchie