TrakHound/MTConnect.NET #51: Message Event seems to appear wrong

benaish, 2023-08-17:
Hi,

My Device.XML contains the following data item:
<DataItem name="XYZ" id="abc" category="EVENT" type="MESSAGE" />

When running an http agent, it appears as:
<Message dataItemId="abc" name="XYZ" sequence="614" timestamp="2023-08-17T07:57:38.6792431Z" nativeCode="Idle "/>

The expected result was:
<Message dataItemId="abc" name="XYZ"  sequence="614" timestamp="2023-08-17T07:57:38.6792431Z">Idle</Message>

Is this a bug, or am I missing something?
I'm using version 5.4.0

Thanks!

PatrickRitchie, 2023-08-18:
Moshe,

Looks like some of your comment is not showing up. See if you can edit and I'll look into what the issue is.

Thanks,
-Patrick

benaish, 2023-08-28:
> Moshe,
> 
> Looks like some of your comment is not showing up. See if you can edit and I'll look into what the issue is.
> 
> Thanks, -Patrick

*Updated*

PatrickRitchie, 2023-08-29:
Is this data coming from a SHDR Adapter? If so, you will need to use the ShdrMessage class instead of the ShdrDataItem class in the Adapter as that will output the correct format for MESSAGE as this has a special NativeCode attribute. If the data is coming from somewhere else (ex. another agent) then this could be an issue that I need to fix.

Thanks,
-Patrick

closed 2024-03-04 by PatrickRitchie