TrakHound/MTConnect.NET #56: Beta 6.0.11 published docker cannot be started

jaxer, 2024-03-04:
Thanks for keeping great work on .NET version of an agent.

I am trying out Beta branch and cannot run the latest docker variant.

Doing:
```
docker pull trakhound/mtconnect.net-agent:6.0.11-beta-alpine-3.18-amd64
docker run trakhound/mtconnect.net-agent:6.0.11-beta-alpine-3.18-amd64
```

That results into:
```
The command could not be loaded, possibly because:
  * You intended to execute a .NET application:
      The application 'agent.dll' does not exist.
  * You intended to execute a .NET SDK command:
      No .NET SDKs were found.

Download a .NET SDK:
https://aka.ms/dotnet/download

Learn about SDK resolution:
https://aka.ms/dotnet/sdk-not-found
```

If I sh into image, then i see no executables in there:
```
#> docker run -it --entrypoint sh trakhound/mtconnect.net-agent:6.0.11-beta-alpine-3.18-amd64
/app # ls -la
total 28
drwxr-xr-x    1 root     root          4096 Feb  2 07:38 .
drwxr-xr-x    1 root     root          4096 Mar  4 12:16 ..
-rwxr-xr-x    1 root     root           355 Feb  2 07:38 Dockerfile
-rwxr-xr-x    1 root     root          1224 Feb  2 07:38 agent.config.default.yaml
drwxr-xr-x    2 root     root          4096 Feb  2 07:38 devices
drwxr-xr-x    2 root     root          4096 Feb  2 07:38 schemas
drwxr-xr-x    2 root     root          4096 Feb  2 07:38 styles
/app # 
```

I am guessing image was built from wrong cwd?

PatrickRitchie, 2024-03-04:
Thanks for letting me know. Yeah it looks like something went wrong during building the docker image. I'll rebuild and publish a new image.

PatrickRitchie, 2024-05-02:
I just published v6.3.2 which should fix the issue. I had something out of order in my build scripts for Docker that weren't copying the files correctly for the past few releases.

Let me know if you find anything else or have any other questions.

jaxer, 2024-05-14:
Thanks for fixing! I tried with latest 6.4.0 and docker image is running as expected. 
I'll close the issue as i was opening it, hope thats the correct process.

closed 2024-05-14 by jaxer