TrakHound/MTConnect.NET #59: Problems with multi-threaded initialization of type ids

SergeyGulik, 2024-04-15:
Hi Patrick

Long time since I wrote you the last time :) 
I see that development is going on. Thank you for your work!

However, I want to complain about minor, but annoying, bug with _typeIds creation
If you go to e.g. DataItem.cs
[image]
you will see that adding new type id is protected by the lock.
Not the case for ConditionObservation.cs, EventObservation.cs, SampleObservation.cs
[image]
Could you please add the same protection for these three and release it in 5.4.x as well?
I am ready to make a PR myself, but I am not sure where is the tip of the stable 5.4.x version.

Best wishes,
Sergii

PatrickRitchie, 2024-04-16:
Thanks for the information and yes I'll fix this issue in the 5.4.x branch. I should have the updated version published tomorrow.

PatrickRitchie, 2024-04-17:
I pushed a new commit that should fix this issue. Let me know if you need an updated Nuget package or compiled agent for v5.4.x. I'm looking to make version 6.x the official release next week.

SergeyGulik, 2024-04-19:
Could you please add it to 5.4.x as well?
We are not in the right moment for migrating to and testing the new 6.x.x
But I will plan it for the future.

commit 0a01564c (2024-04-17): Update to fix #59
closed 2024-05-16 by PatrickRitchie