TrakHound/MTConnect.NET #61: Add Multiple Devices in Embedded MTConnect Agent

smtpl-iiotian, 2024-04-16:
@PatrickRitchie this is another great development of embedded mtconnect agent. I tested it for 1 machine and it worked as expected. I was able to route Mitsubishi CNC data to MTConnect agent. But I was only able to add 1 device in the protected override IDevice OnAddDevice() function. 

Please guide how can I add x number of devices.

PatrickRitchie, 2024-04-16:
Glad to hear you have found the embedded agent helpful. I intended for additional devices to be added by adding multiple configurations to the configuration file similar to below:

```yaml
- haas-ethernet:
    deviceKey: VF2-01
    server: 192.168.1.101
    port: 5051

- haas-ethernet:
    deviceKey: VF2-02
    server: 192.168.1.102
    port: 5051

- haas-ethernet:
    deviceKey: VF2-03
    server: 192.168.1.103
    port: 5051
```

This will create an instance of the same module for each configuration.

Let me know if this works for you or if you have any other questions or recommendations.

smtpl-iiotian, 2024-04-19:
@PatrickRitchie very much thanks. I was able to add x number of devices after configuration update.

smtpl-iiotian, 2024-04-19:
@PatrickRitchie is there any provision in the embedded mtconnect agent to make the executable .exe as a windows service OR I have to do it manually?

smtpl-iiotian, 2024-05-14:
@PatrickRitchie, I tried to install published .exe as a windows service as per the above link and guide. 
I added serviceName in configuration file to add multiple windows services one for Mitsubishi and another for Modbus devices but it only gets applied as a Display name as seen in the Windows Process(image attached). 
[image]

The windows Service name remains the same as MTConnect.Net-Agent. So when I try to install Modbus.exe with admin privileges of cmd the earlier windows service Mitsubishi-MTConnect.Net-Agent gets replaced with Modbus-MTConnect.Net-Agent

[image]
Attached snippet of configuration file

Kindly guide, If I am going wrong

PatrickRitchie, 2024-05-17:
See if this is fixed in the [v6.4.1 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v6.4.1). 

I added serviceDisplayName and serviceDescription in the configuration file:
```yaml
serviceName: Mitsubishi-MTConnect.NET-Agnet
serviceDisplayName: Mitsubishi MTConnect.NET Agent
serviceDescription: MTConnect Agent to read information from a Mitsibishi Machine
```

I also fixed an issue with passing a configuration file path when installing the agent as a service

Please let me know if this is still an issue or if you find anything else.

koturcu, 2024-06-06:
@smtpl-iiotian Hello, I am trying to data collect  from a Mitsubishi CNC machine and I am writing a thesis on this subject, can you help me? It was written on some sites that free data cannot be collected from Mitsubishi, is this true?

smtpl-iiotian, 2024-06-12:
@PatrickRitchie apologies for late reply. It is working now.

smtpl-iiotian, 2024-06-12:
@koturcu explore NC Communication SDK on https://www.mitsubishielectric.com/fa/download/software/search.page?mode=software&kisyu=/cnc
It is not free as said.

koturcu, 2024-06-12:
Thank you sir