TrakHound/MTConnect.NET #63: AgentConfiguration.Read ignores path argument

victoririzarx3d, 2024-05-16:
Hi @PatrickRitchie 

We are testing now the agent implementation on v6.4.0 and we are realizing that the agent is ignoring the configuration path we are providing via the CLI arguments and always falling back to the default configuration

We are testing on Ubuntu 20.04, and built the v6.4.0 branch using `dotnet publish -c:Debug -r:linux-x64 -f:net8.0 -o:dist --sc`

We run the agent with `./mtconnect_dotnet_agent  /home/user/data/state/mt_connect/agent.config.yaml`

The custom config on the path we specify is a copy of the default agent configuration with a couple of minor changes

And this is the log output we get, Notice how it chooses the default configuration on `/usr/local/mtconnect_dotnet_agent/agent.config.yaml` although we would expect to use the one passed in the CLI argument

```
--------------------
Copyright 2024 TrakHound Inc., All Rights Reserved
MTConnect.NET Agent : Version 6.4.0.0
--------------------
This application is licensed under the MIT License (https://choosealicense.com/licenses/mit/)
Source code available at Github.com (https://github.com/TrakHound/MTConnect.NET)
--------------------
2024-05-16 02:46:56.4109|INFO|application|Agent Configuration Path = /home/user/data/state/mt_connect/agent.config.yaml
2024-05-16 02:46:56.4979|INFO|application|Configuration File Read Successfully from: /usr/local/mtconnect_dotnet_agent/agent.config.yaml

```

Looking a bit deeper into the code, the MTConnectAgentAppllication uses the AgentConfiguration.Read method to read the configuration file passed from the CLI arguments 

https://github.com/TrakHound/MTConnect.NET/blob/cf3a28994e690bad821b03e78bc16a073952b476/agent/MTConnect.NET-Applications-Agents/MTConnectAgentApplication.cs#L92-L103

https://github.com/TrakHound/MTConnect.NET/blob/cf3a28994e690bad821b03e78bc16a073952b476/agent/MTConnect.NET-Applications-Agents/MTConnectAgentApplication.cs#L136-L142

https://github.com/TrakHound/MTConnect.NET/blob/cf3a28994e690bad821b03e78bc16a073952b476/agent/MTConnect.NET-Applications-Agents/MTConnectAgentApplication.cs#L305-L309

The AgentConfiguration.Read method, however, ignores the passed path and always constructs a default configuration file

https://github.com/TrakHound/MTConnect.NET/blob/cf3a28994e690bad821b03e78bc16a073952b476/libraries/MTConnect.NET-Common/Configurations/AgentConfiguration.cs#L138-L164

PatrickRitchie, 2024-05-16:
Thank you for the information. Yes it looks like the AgentConfiguration.Read() method is ignoring the path parameter.

I will publish a new version with this fixed.

Please let me know if you find any other issues.

PatrickRitchie, 2024-05-17:
See if this is fixed in the [v6.4.1 Release](https://github.com/TrakHound/MTConnect.NET/releases/tag/v6.4.1). Please let me know if this is still an issue or if you find anything else.

victoririzarx3d, 2024-05-21:
@PatrickRitchie tested on v6.4.1 and working fine now, thanks. I will keep you updated if I find something else.

closed 2024-05-21 by victoririzarx3d