TrakHound/MTConnect.NET #65: Module http-adapter not loaded

anirbanSP, 2024-06-14:
---

### Issue: `http-adapter` Module Not Loading

#### Description
Hi,

I have installed the MTConnect agent from the .exe file on my Windows 10 PC. I am trying to launch the .exe with the following configuration:

```yaml
modules:
  - http-adapter:
      address: mtconnect.mazakcorp.com
      port: 5610
      deviceKey: M12346
      interval: 100
      heartbeat: 10000
      useSSL: false
      currentOnly: false
      useStreaming: false
      
  - mqtt-relay:
      server: localhost
      port: 1883
      topicPrefix: MTConnect
      topicStructure: Document
      documentFormat: JSON-CPPAGENT-MQTT
      currentInterval: 5000
      sampleInterval: 5000
      clientId: my-mqtt-client
      reconnectInterval: 10000
      currentOnly: false
```

But the agent logs show that it is not loading the `http-adapter` module for some reason. Here is the log:

```
2024-06-14 11:46:14.9541|INFO|modules.mqtt-relay|MQTT Relay Connected to External Broker (localhost:1883)
2024-06-14 11:49:28.9510|INFO|application-logger|[Application] : Agent Configuration File Updated (C:\Program Files\TrakHound\MTConnect.NET-Agent\agent.config.yaml)
```

Is there something wrong with the configuration (because it's copied from the example on GitHub)?

#### Steps to Reproduce
1. Install the MTConnect agent using the .exe file on Windows 10.
2. Use the provided configuration.
3. Start the agent and observe the logs.

#### Expected Behavior
The `http-adapter` module should load and connect to the specified address and port.

#### Actual Behavior
The `http-adapter` module does not load, while the `mqtt-relay` module connects successfully.

#### Environment
- Windows 10
- MTConnect.NET Agent version: 6.4.1.0
---

anirbanSP, 2024-06-14:
the "http-adapter" module should be loaded as "http-client" in the config file. I think the example configuration for http-adapter should be updated to reflect this change.

closed 2024-06-14 by anirbanSP

PatrickRitchie, 2024-06-14:
Thank you for finding this issue. It looks like this was a mistake in the module and not the example configuration. I just published a new release that uses the 'http-adapter' ID. Please let me know if you have any further questions/thoughts about this.

[Release 6.4.3](https://github.com/TrakHound/MTConnect.NET/releases/tag/v6.4.3)