TrakHound/MTConnect.NET #68: config files not published

smtpl-iiotian, 2024-07-03:
@PatrickRitchie in the latest build 6.4.3, agent.config.default.json, agent.information.json i.e., basically config files are not getting publised after building or publishing the application. 

I had to add those files manually from old 6.2.2 published application.

PatrickRitchie, 2024-07-16:
The 'agent.information.json' file is created at runtime and is mainly used to keep the agent UUID consistent for the install (if you install a new version, the UUID remains the same). It also contains some information for when the 'durable: true' option is used. This file should not typically need to be copied when building the source code or from one installation to another.

In version 6, I am looking to use YAML instead of JSON for configuration files. There is still some compatibility to read older JSON config files but new config files should be written using YAML. The "default" configuration file is now 'agent.config.default.yaml' and should be included in the MTConnect.NET-Applications-Agents project.

Below is an example of a YAML config file:
```yaml
# - Device Configuration -
devices: devices

# - Processors -
processors:
- python: # - Add Python Processor
    directory: processors

# - Modules -
modules:
  
- http-server: # - Add HTTP Server module
    hostname: localhost
    port: 5000
    allowPut: true
    indentOutput: true
    documentFormat: xml
    responseCompression:
    - gzip
    - br
    files:
    - path: schemas
    location: schemas
    - path: styles
    location: styles
    - path: styles/favicon.ico
    location: favicon.ico

- mqtt-relay: # - Add MQTT Relay module
    server: localhost
    port: 1883
    currentInterval: 5000
    sampleInterval: 500

- shdr-adapter: # - Add SHDR Adapter module for Device = M12346 and Port = 7878
    deviceKey: M12346
    hostname: localhost
    port: 7878

- shdr-adapter: # - Add SHDR Adapter module for Device = OKUMA-Lathe and Port = 7879
    deviceKey: OKUMA-Lathe
    hostname: localhost
    port: 7879

- mqtt-adapter: # - Add MQTT Adapter module for Device = M12346 and Topic = cnc-01
    deviceKey: M12346
    server: localhost
    port: 1883
    topic: cnc-01

# The maximum number of Observations the agent can hold in its buffer
observationBufferSize: 150000

# The maximum number of assets the agent can hold in its buffer
assetBufferSize: 1000

# Sets whether the Agent buffers are durable and retain state after restart
durable: false

# Sets the default MTConnect version to output response documents for.
defaultVersion: 2.3
```

Please let me know if you have any questions about this or need any assistance with converting JSON to YAML.

smtpl-iiotian, 2024-08-16:
@PatrickRitchie, I checked with the latest v6.4.0 but the default file 'agent.config.default.yaml' is not seen in the project after installing MTConnect.NET-Applications-Agents library from nuget package. 
But installing the earlier v6.2.2beta from the nuget package made default agent.config.default.yaml file visible in the project.
Please check.