TrakHound/MTConnect.NET #85: Retain Flag Not Honored by MTConnect Agent

Pranay265, 2024-10-30:
It appears that the retain flag in MQTT messages is not being honored by the MTConnect agent. When enabled, the retain flag should keep the last known good message for any new subscribers. however, messages from the agent are not retained as expected.

MQTT Relay config:

`- mqtt-relay: # - Add MQTT Relay module (TLS)
    QoS: 1
    server: localhost
    port: 8883
    clientId: print123
    currentInterval: 10000
    sampleInterval: 500
    allowUntrustedCertificates: true
    documentFormat: JSON
    topicPrefix: MTConnect
    topicStructure: Entity
    tls:
      pem:
        certificateAuthority: certificates/GroupCACertificate.pem
        certificatePath: certificates/certificate.pem
        privateKeyPath: certificates/privateKey.pem
      verifyClientCertificate: false
      omitCAValidation: true`

**Broker:**
The broker is set up to retain messages, and configuration appears as expected. However, only active data items are visible, not retained ones.

[image]

It’s set as below, but it’s unclear why it’s not getting applied. Am I missing anything in the configuration?

`  private async void ProbeReceived(IDevice device, IDevicesResponseDocument responseDocument)
  {
      if (_mqttClient != null && _mqttClient.IsConnected)
      {
          var x = new List<KeyValuePair<string, string>>();
          x.Add(new KeyValuePair<string, string>("indentOutput", _configuration.IndentOutput.ToString()));

          var formatResult = ResponseDocumentFormatter.Format(_configuration.DocumentFormat, responseDocument, x);
          if (formatResult.Success)
          {
              var topic = $"{_configuration.TopicPrefix}/{MTConnectMqttDocumentServer.ProbeTopic}/{device.Uuid}";
              if (formatResult.Content != null && formatResult.Content.Position > 0) formatResult.Content.Seek(0, SeekOrigin.Begin);

              try
              {
                  var messageBuilder = new MqttApplicationMessageBuilder();
                  messageBuilder.WithRetainFlag(true);
                  messageBuilder.WithTopic(topic);
                  messageBuilder.WithQualityOfServiceLevel(MQTTnet.Protocol.MqttQualityOfServiceLevel.AtLeastOnce);
                  messageBuilder.WithPayload(formatResult.Content);
                  var message = messageBuilder.Build();
`
Please let me know if any additional information needed from my end.

Pranay265, 2024-12-05:
Hello @PatrickRitchie 
I wanted to follow up on this issue as it’s been a while since the last comment. Is there any progress or anything I can do to help move things forward? Please let me know if you need more details or if there’s anything I can assist with. Looking forward to hearing from you. Thanks!

PatrickRitchie, 2024-12-20:
Are you still seeing the same issue? It looks like I'm able to see the retained flag using v6.5.1 as shown below:

[image]

Pranay265, 2025-01-07:
Hi @PatrickRitchie,

Apologies for the delayed response. Yes, I can now see the retain flag as expected. Thank you so much for your help in resolving this issue.

closed 2025-01-07 by Pranay265