TrakHound/MTConnect.NET #89: MQTT Relay Connection Error : Error while authenticating. Connection closed.

AlexanderIoffe, 2024-12-17:
Hello,
I want to update Agent and Adapter from 5.4.1 to 6.5.0 and faced problem in 6.5.0 while i run Agent as Process from code and
get an exception 
**2024-12-17 12:10:41.9189|0|WARN|modules.mqtt-relay|MQTT Relay Connection Error : Error while authenticating. Connection closed. 
2024-12-17 12:10:41.9295|0|WARN|modules.mqtt-relay|MQTT Relay Connection Error (InnerException) : Connection closed. 
2024-12-17 12:10:41.9295|0|WARN|modules.mqtt-relay|MQTT Relay Connection Error (BaseException) : MQTTnet.Exceptions.MqttCommunicationException: Connection closed.
   at MQTTnet.Client.MqttClient.Authenticate(IMqttChannelAdapter channelAdapter, MqttClientOptions options, CancellationToken cancellationToken)** 

When run Agent by click to exe it works fine.

In version 5.4.0 it works it both scenario.

Thank you

PatrickRitchie, 2024-12-20:
The certificate paths you have set in the config file are relative so if you are running it as a process from code, you may need to set the working directory using Process.StartInfo. Let me know if you are still having issues with this.

AlexanderIoffe, 2024-12-25:
Thank you @PatrickRitchie .
I changed the agent config and it's working.

closed 2025-01-18 by PatrickRitchie