TrakHound/MTConnect.NET #107: Publishing MQTT from Agent

MRIIOT, 2025-09-08:
Below code embeds `MTConnectAgentApplication` and serves up an Agent on http/5000.  This works perfect.  Can you point me to classes or examples that would enable publishing Probe/Current/Sample to an MQTT broker instead of serving over HTTP?

```
public class Sink: SinkConnector<ConnectorConfiguration, ConnectorItem>
{
    private MTConnectAgentApplication _client = null;
    
    protected override bool CreateImplementation()
    {
        _client = new MTConnectAgentApplication();
        return true;
    }

    protected override bool ConnectImplementation()
    {
        _client.StartAgent(new AgentApplicationConfiguration() { 
            Modules = new []
            {
                new Dictionary<object, object> ()
                {
                    {"http-server", new Dictionary<object, object>()
                    {
                        {"port", Configuration.Port},
                    }}
                }
            }
        });
        
        return true;
    }

    protected override bool WriteImplementation()
    {
        foreach (var message in Outbox)
        {
            if (message.ConnectorItemRef is not null &&
                message.ConnectorItemRef.SinkMeta is not null &&
                message.ConnectorItemRef.SinkMeta.ContainsKey("mtconnect"))
            {
                var devices = _client.Agent.GetDevices().ToDictionary(o => o.Name, o => (Device)o);
                
                var (wasModified, device, dataItem) = 
                    Builder.Build(
                        devices, 
                        message.ConnectorItemRef.SinkMeta["mtconnect"].ToString(), 
                        message.Path);

                if (wasModified)
                {
                    _client.Agent.AddDevice(device);
                }
                
                _client.Agent.AddObservation(
                    device.Uuid, 
                    dataItem.Id, 
                    Configuration.UseSinkTransform ? TransformAndSerializeMessage(message) : message.Data, 
                    DateTimeOffset.FromUnixTimeMilliseconds(message.Timestamp).LocalDateTime);
            }
        }
        
        return true;
    }

    protected override bool DisconnectImplementation()
    {
        _client.StopAgent();
        return true;
    }
}
```

MRIIOT, 2025-09-09:
Should the MTConnectAgentApplication's Agent property be MTConnectAgentBroker type?

[image]

PatrickRitchie, 2025-09-10:
Let me know if you are still having issues. It should be fine to read from the Configuration object to set the parameters in the Dictionary.

MRIIOT, 2025-09-10:
I kind of figured it out.... To get the MQTT-Relay working in DIME I had to explicitly change every MTConnect.NET* project's target framework to 9.0.  Then I built them, copied them into the output folder of my DIME project and it works. Strange.  Do you want to get on a call tomorrow or something, I can show you.

closed 2025-09-16 by MRIIOT